shiv3/ocpp-cp-simulator #7: Charge point basic authentication

philippbruhin, 2024-09-27:
Dear @shiv3, Thank you for sharing this valuable project with the community!

I noticed that it's possible to add the WebSocket URL under the settings. Is there a plan to support OCPP passwords as well? I couldn't find any implementation details regarding this in the [ts-ocpp repository](https://github.com/voltbras/ts-ocpp). However, OCPP defines basic authentication as option: 

[image]
*Open Charge Point Protocol JSON 1.6, OCPP-J 1.6 Specification*

shiv3, 2024-09-28:
@philippbruhin 
Hi, thank you for finding this repository and giving feedback.
I tried implementing Basic Auth settings in #8 , but I think there’s a high possibility that it won’t work.
because this OCPP charger simulator is designed to run in the browser, but as mentioned in the StackOverflow post below, it seems difficult to attach Authorization headers to WebSocket requests from the browser.
so please try connecting with the implementation in #8, and if it seems difficult to get it working, it might be better to choose a solution that works outside the browser.
https://stackoverflow.com/questions/4361173/http-headers-in-websockets-client-api

shiv3, 2024-09-28:
this repository does not directly use the ts-ocpp ws implementation, but I think it will be usable if you use the setting that I added
https://github.com/voltbras/ts-ocpp/pull/56

philippbruhin, 2024-09-28:
Hi @shiv3 ,

Thank you so much for implementing this feature! I really appreciate your effort. Unfortunately, as you anticipated, it didn't work on my end. Please see following screenshot for reference:

[image]

Thanks to the links you provided my summarized learnings are:

**1. Browser Limitation:**
Browsers do not allow the `wss://username:password@...` format for WebSocket connections due to security concerns. Passing credentials in the URL is blocked by most modern browsers for security reasons, and WebSocket URLs don’t support this format.

**2. OCPP Server Expectation (Header vs URL):**

OCPP servers expect the Authorization header, not credentials embedded in the WebSocket URL. This is because:

- **OCPP specification:** According to the OCPP protocol and WebSocket standards, authentication (if Basic Auth is used) is typically handled via HTTP headers, especially during the WebSocket handshake (before the connection is upgraded). The server expects to receive an Authorization header with `Basic xxxx...` credentials as part of the initial WebSocket connection's HTTP handshake.

- **Credentials in URL are non-standard:** Embedding credentials in the WebSocket URL (wss://username:password@...) is not part of the OCPP standard for authentication, nor is it part of the WebSocket protocol’s secure practices. The server won’t check for credentials in the URL because it looks specifically for them in the headers, as per the OCPP (or any WebSocket-based) specification.

I’ll look into a solution that works outside of the browser, for example https://github.com/matth-x/MicroOcppSimulator. Feel free to close this issue.

By the way, do you work a lot with OCPP? I’d love to stay in touch and exchange ideas, as I am working on similar projects. Is there any way to contact you? 

Best regards,
Philipp

shiv3, 2024-10-06:
@philippbruhin
Thank you for your appreciation and for reporting the results of your extensive research.
I thought that was probably the case. I'll revert the fix I added. Thank you for checking!
Also, this emulator that runs outside of a browser might be useful( it was shared in reddit). Check it out.
https://ocpp-emulator.monta.com/

Yes, I usually use OCPP to develop EV charger systems (as a side job tho).
If you'd like, you can contact me at x( [lailaiskyn](https://x.com/lailaiskyn) )

closed 2024-10-06 by shiv3